awolf78/KISS_OSD #1: Yaw stick movrement

fedorcomander, 2016-10-13:
While disarmed, move yaw to the right. Now you can select the batteries with roll and change the values with pitch.

How does it work for people with ARM-ON-YAW ?

awolf78, 2016-10-13:
I actually did try this since I only have a 6-channel radio. It doesn't work. I'm guessing Felix deactivated the feature. And I'm glad he did: Think about punching up and then throttle to min to get maximum hang time and then yaw like crazy - and then your copter disarms???
Sorry, but this is unsafe!!!

fedorcomander, 2016-10-13:
Its actually works. I just checked. Arms and rearms with yaw command as usual. That why you need  min_command... And there ARE people who actually using it, btw. ;)

awolf78, 2016-10-13:
How about this:
If there is no ARM switch configured for AUX1-AUX4, I can assume the user is using ARM on yaw. I should be able to read this off the settings, which I just need to ask for a little earlier. Then I just disable the battery change feature on yaw right. Instead, the user needs to go into the menu to change his battery. I would add an extra entry.

Now you have a chance to be EXTRA helpful today :)
You could tell me where to look for in the settings buffer :)

fedorcomander, 2016-10-13:
Check my gui stuff. Its all there.

awolf78, 2016-10-13:
Looking at:

data.setUint8(73, obj.aux1Funk);
data.setUint8(74, obj.aux2Funk);
data.setUint8(75, obj.aux3Funk);
data.setUint8(76, obj.aux4Funk);

So byte 73-76 contain the AUX1-4 settings. I am assuming:
No Action = 0
High Arm = 1
Mid Arm = 2
Low Arm = 3

Correct?

fedorcomander, 2016-10-13:
Not exactly.

awolf78, 2016-10-13:
Ok, let's try the magic word: Please?

awolf78, 2016-10-13:
Never mind, I found it:

<option value="1">High: ARM</option>
<option value="12">Mid: ARM</option>
<option value="13">Low: ARM</option>

awolf78, 2016-10-13:
Never mind, I found it. 1, 12 and 13.

awolf78, 2016-10-13:
tested and updated in master

closed 2016-10-13 by awolf78