awolf78/KISS_OSD #2: Working with blackbox enabled

fedorcomander, 2016-10-13:
There WILL garbage on the screen when BB is connected to the FC and operational. I sent fix to felix, and he supposed to put it to his git somewhere ;)))

fedorcomander, 2016-10-13:
Thats from new code ;)

 if(micros()-LastLoopTime > 10000){
    LastLoopTime = micros();
  
    Serial.write(0x20); // request telemetrie
    
    minBytes = 100;
    recBytes = 0;
    packetReceived = 0;
   
    while(recBytes < minBytes && micros()-LastLoopTime < 50000){
      #define STARTCOUNT 2
      if(Serial.available()) serialBuf[recBytes++] = Serial.read();
      if(recBytes == 1 && serialBuf[0] != 5)recBytes = 0; // check for start byte, reset if its wrong
      if(recBytes == 2) {
          minBytes = serialBuf[1]+STARTCOUNT+1; // got the transmission length
          if (minBytes<150 || minBytes>180)  {
              recBytes = 0;
              minBytes = 100;
          }
      }
      if(recBytes == minBytes){
         uint32_t checksum = 0;
         for(i=2;i<minBytes;i++){
            checksum += serialBuf[i];
         }
         checksum = (uint32_t)checksum/(minBytes-3);

awolf78, 2016-10-13:
So the key is:
if (minBytes<150 || minBytes>180)  {
          recBytes = 0;
          minBytes = 100;
      }

This is ok as a short term fix, I would actually just use a return inside the if statement. Then the OSD would not show anything - makes more sense to me since it won't work anyhow when the logger is active. if minBytes ever grows beyond 180 bytes, the OSD won't work any more. 
It would be much cleaner to detect the cleanflight protocol and either parse the useful data into the OSD or turn it off. Then you would need to be able to change the baud rate in the menu. Not sure if I will waste precious program code space for this - only about 4K left on the NVRAM.

Updated in master.

closed 2016-10-13 by awolf78