awolf78/KISS_OSD #5: Left side of the OSD off screen.

RobbieHood, 2016-12-07:
Hello,

I flashed this software to my micro minimOSD.
The left side off the OSD is not on my screen.

What can I do ?

awolf78, 2016-12-07:
What type of goggles are you using?
Is there any adjustment on your goggles to move the picture?

maix0000, 2016-12-13:
I have the same issue. The left part of the OSD seems to be cut off.

I am using:
- HS1177 FPV Camera
- KISS FC, KISS 24RE
- PAL version of your OSD
- Fatshark HD2 Goggles

Any help is appreciated. Great OSD BTW.

Cheers, Stefan

awolf78, 2016-12-14:
I cannot really test this since I don't have a PAL camera.
Can you do me a favor and change line 134 in KISS_OSD.ino from
static const uint8_t COLS = MAX7456_COLS_P1;

to 

static const uint8_t COLS = MAX7456_COLS_P0;

If this makes it worse try:

static const uint8_t COLS = MAX7456_COLS_P2;

Not sure if there is an adjustment on your Fatshark goggles - I'm using Headplay.

maix0000, 2016-12-14:
Thank you, I tested both values in line 134 "P0" and "P2". Alas, there was no improvment.
Anyway, I replaced the HS1177 cam with a Runcam Eagle which can be switched to NTSC. And voila now it works flawlessly.
I will test the OSD and put my feedback in the youtube thread.
Thx again, Stefan

awolf78, 2016-12-14:
Thank you very much for testing.
Looks like I will have to add some padding for the PAL versions :(

awolf78, 2016-12-14:
You can try checking out the latest master and try that. You should (but don't have to) update the font as well - otherwise you cannot see the new ESC symbol

maix0000, 2016-12-14:
OK, great. I’ll check and revert.

Cheers,
Stefan

awolf78, 2016-12-17:
I think I found the culprit. There was an offset from Felix' original OSD code.
You can either download the latest master (make sure you update font and change to PAL) or download the latest PAL release.
I'm keeping this open for a while, but if I don't hear anything in the near future - I'm closing this.

maix0000, 2016-12-17:
Thank you. I guess you can close the issue. Like said (posted) before I have moved to the NTSC version. I will download the latest master and test it, alas not before Monday 19, December.
Thx again and best regards,

Stefan

PS: Keep it up, I`d love to see PID tuning via OSD.

closed 2016-12-19 by awolf78