awolf78/KISS_OSD #7: ImmersionRC Tramp

leisim, 2017-01-25:
Does this OSD support setting band and power of the Tramp vTx? ;)

awolf78, 2017-01-25:
Tramp vTx:
Fedorcomander is working on an implementation so you can change channels via the KISS FC GUI. Implementing this on the OSD is unfortunately not possible, as this would require an additional serial port, which is not available on current OSD hardware. Theoretically it would be possible to change channels on the Tramp via the KISS FC - but this would require two things:

- Flyduino has to officially release the FW so you can write settings while a lipo is connected (they say this will happen with V1.2)

- The Tramp or the OSD would have to be connected to one of the additional serial ports. I know in Fedor's current implementation that is not supported.

KISS OSD CONFIG tool:
I wish I could put it all in one so you don't have to flash two different FWs. The issue is space: The ATMEL328 chip does not have enough memory. MWOSD uses an external software for configuration. I thought: if you have to connect the FTDI anyhow, you might as well put it on the OSD - then it is WYSIWYG: What You See Is What You Get.

leisim, 2017-01-27:
Thanks for the detailed answer ;)

The config tool is a good idea but I hope they support multiple serial ports at the future.

fedorcomander, 2017-01-27:
Well... I might move VTX control to PWM5 or PWM6 (which is a bummer on AIO boards)... But for now, i NEED TX to be connected to VTX... Which makes OSD life not easy at all...

awolf78, 2017-01-27:
That's a pain, then you have to use a software serial. 
Is there no way we can use RX2 or RX3?
[https://www.google.com/url?sa=i&rct=j&q=&esrc=s&source=images&cd=&cad=rja&uact=8&ved=0ahUKEwiagJXv4OLRAhWLJSwKHby6Ac0QjRwIBw&url=https%3A%2F%2Foscarliang.com%2Fkiss-fc-flight-controller%2F&psig=AFQjCNEY_98Pf81LVRI0UGXLBWRBdvMguw&ust=1485621259956227](url)

fedorcomander, 2017-01-27:
Well... I added you commands to control vtx ;) Would be nice to have it implemented ;) The rest it uo to FC what to connect to.

awolf78, 2017-01-27:
Will add vTx for sure.
Any chance you can include TPA (to 0x12) as well?

I will check with Felix if we can make use of the other serial ports.

fedorcomander, 2017-01-27:
Other serial ports are being used. 

I dont think tpa belongs here. Otherwise we are ending up with same configuration monstrosity.

awolf78, 2017-01-27:
Too bad.
Anyhow, I'll get this implemented and tested over the weekend. Thanks for all your help!

TPA: I just wanted to add in the "simple" TPA, not the custom one. Only a few extra bytes and it is definitely part of the tuning. I often have to fiddle with TPA once I get close to max P.

awolf78, 2017-01-30:
Time to close this.
@fedorcomander Many thanks for all your help and support!

Tramp control is implemented on OSD, now all that needs to be done is a software serial for PWM5 or 6 on KISS FC - but OSD wise we are done.

closed 2017-01-30 by awolf78

leisim, 2017-01-30:
Wow that is great news. Thank you very much. You guys are awesome!!!