awolf78/KISS_OSD #9: KISS OSD appears briefly then disappears when ESCs initialise

Bens1606, 2017-02-23:
Hi, I wonder if you can help me? I've set up a Kiss OSD exactly as shown on your page but when I connect the battery, the OSD appears in my goggles for a few seconds but then disappears as soon as the ESCs initialise (beep). I'm running with Kiss 24A ESCs. Kiss FC firmware is 1.1RC6. Logger is switched off in the Kiss configurator. Any ideas?

Bens1606, 2017-02-23:
Update on the above...I've just been playing with the settings in the configurator and if I set it to Oneshot 42, the OSD stays on, so it seems to be a problem with Dshot? This seemed like good news but now I can't get into the setup menu 😁 I've tried yawing left for 5 seconds but nothing happens. Any ideas what's going on and do you know if there is a fix to allow me to use Dshot?

awolf78, 2017-02-23:
Let's continue this on my rc-groups thread, you can upload some pictures there (not possible on github):
https://www.rcgroups.com/forums/showthread.php?2806289-KISS-OSD-Firmware-new-versions-posted-here

Can you post some pictures about the wiring of your OSD? 
Most OSD hardware (minimOSD, piggyOSD) is quite sensitive to voltage spikes which causes a loss of video overlay. Need to see how you have wired it up to help you further.

Bens1606, 2017-02-24:
Hi and thanks for your reply. I'll get some pictures posted over the weekend. 

Many thanks

Ben

Bens1606, 2017-02-24:
Hi again, 

I've just read through your Rcgroups thread and I think I might have misunderstood the sales spiel on the flyduino website? When I ordered my OSD off their site, I assumed that it came with your firmware as it shows your videos on their page. It says that the OSD comes fully preflashed and so no need to flash new firmware. Have I misunderstood? If so apologies, I'll take this issue up with the flyduino support team.

Regards 

Ben

awolf78, 2017-02-24:
The KISS OSD comes pre-flashed with KISS OSD 1.0. I took the code from 1.0 and added a lot of functionality to it which then became 2.x (currently at 2.2).
Even if you don't use my FW, I would still be happy to help you.

Bens1606, 2017-02-24:
Thanks very much, very good of you. I was hoping to get away without flashing any firmware but your firmware looks really good so I think I'm going to order myself an FTDI adapter and get the OSD up to your latest fw. Depending on how it goes, I might be in touch! 

I think the problems I'm having are to do with DShot as it seems to work on Oneshot42. At least I can see the OSD but I can't seem to do anything with it. Do you know how I can get into the setup menu on the fw 1.0?

Thanks for your help (so far)

Regards

Ben

awolf78, 2017-02-24:
V1.0 has no menu. I added that V2.x. V1.0 is very basic, you cannot change anything, it just displays the data - that's it.
If you ever update to V2.x - check out  my Wiki on the github page and contact me if you have any issues.

Bens1606, 2017-02-24:
That's great thanks very much

closed 2017-03-04 by awolf78

Bens1606, 2017-03-25:
Hi,

Sorry to bother you again but I wonder if you can help? I installed your firmware, (which is awesome by the way) and finally got it running on the bench. I took it for a first flight today (so cold!) but it just cuts out about 30seconds into the flight and then the video feed is very dodgy with lines across the screen and break ups etc. I have a 470uf cap soldered across the power in, as per your reccomendation which fixed my Dshot problem (see below) so it shouldn't be voltage spikes. 

Have you come across this before?

Many thanks

Ben

awolf78, 2017-03-25:
Sounds like a ground wire issue to me.
Check out the first sketch in my wiki:
[https://github.com/awolf78/KISS_OSD/wiki](url)

Make sure the ground wires are all connected as described. Otherwise you might introduce a ground loop.
Not sure what gear you are using, but if you use a minimOSD (or the Flyduino one, that's the same) and a vTx providing 5V the safest way is to power the OSD and the camera from the 5V provided by the vTx (and use the same ground).

Bens1606, 2017-03-26:
Hi 

Thanks again for your help. I've checked all the ground wires and they are wired up as per your diagram and I've checked that there are no short anywhere. I have it powered from a 5v Pololu, not from the Vtx. Do you think it will make a difference to try powering it from the Vtx?

Cheers

Ben

awolf78, 2017-03-26:
Pololu can introduce a ground loop - depends how it is wired. I would try using the 5V from the vTx.

Bens1606, 2017-03-28:
Ok thanks, I'll try connecting it that way. 

Thanks for your help.

Ben

Bens1606, 2017-04-01:
Hi again,

Just wanted to say thanks for your help and advice. Finally got it hooked up with power from the Vtx and it's working great at last.  Am loving being able to see the batt voltage and flight time on the screen....takes out a lot of the guess work.

Video signal is a bit noisy (horizontal lines) so I've got a bit of work to do to isolate the cause but should be solvable I think.

Now just looking forward to some decent weather to get out and fly!

Thanks again and all the best

Ben

awolf78, 2017-04-01:
Hi Ben,

I'm guessing you are using a Unify Pro HV as vTx? I had the same issue. 
Solution1: Solder on big capacitor.
Solution2: Complain with TBS and send it to HongKong for a replacement.

Check out this thread:
https://www.rcgroups.com/forums/showpost.php?p=35108031&postcount=1

Good luck!

Alex

Bens1606, 2017-04-01:
Hi Alex,

I'm actually using a Foxeer TM25. The interference isn't that bad but is definitely motor/esc generated. I've tried rerouting and twisting the cables but no change so I think I probably need to fit an LC filter. I'll give it a go and see what happens.

All best

Ben