awolf78/KISS_OSD #10: TBS SmartAudio

jasonbenassi, 2017-03-21:
Hello,

Thanks so much for producing this OSD, it works well! Do you have any plans to incorporate TBS SmartAudio controls? It would be awesome to manage TBS Unify Pro VTX settings with the sticks.

Cheers,
Jason

awolf78, 2017-03-21:
Due to a lack of serial ports on the KISS FC side this will take a bit longer. KISS FC has only one free serial port which is used for either SmartAudio or OSD. But in the future it is planned to use one of the PWM ports for Tramp/SmartAudio - not sure if that will make it in the upcoming release though. Thank you for your patience.

aschneider, 2017-05-02:
Smart Audio will go on RX3 Port on the KISS FC in order to workaround this issue as stated by FedorCommander on RC Groups. So this issue can be closed once new FW is available...

awolf78, 2017-05-08:
I don't have any control when KISS FC FW 1.3 will be released, but it will be everyone's wet dream when it comes to vTx control and OSD :)
So I'm taking the liberty and close this now.

closed 2017-05-08 by awolf78

orionpt, 2017-06-03:
I'm planning on building my first full KISS setup and want to use a TBS UPro. Just found this post and wanted to leave you a tip to allow setting user custom frequency. It would be awesome to have that feature incorporated.

awolf78, 2017-06-03:
The OSD is just sending commands to the KISS FC which actually controls the vTx. I do not have access to the KISS FC code, but you are welcome to ask Fedorcommander in the rcgroups forum about this feature.

fedorcomander, 2017-06-03:
not gonna happen. but buddies of mine hack freq table in firmware hex file and all works just fine. 

---
Cheers,
Alex

orionpt, 2017-06-03:
That frequency hack is done on the TBS firmware, right?

fedorcomander, 2017-06-03:
on kiss or bf. 

---
Cheers,
Alex