awolf78/KISS_OSD #17: Problem when starting the firmware

xXtrem, 2017-04-04:
Hi,
in advance: Nice work, many thanks.
My problem with 2.3:
After starting the osd_config tool, the menu will appear for 2 seconds and then disappears completely.
The osd tool does not run at all, no menu, no response.
Version 2.2. works for me as expected.

Greetings Frank

awolf78, 2017-04-04:
How did you update the OSD? With Arduino? Or XLoader?
If Arduino: What version did you use?

awolf78, 2017-04-04:
Another question: What camera do you use? Is it a NTSC or PAL camera?

xXtrem, 2017-04-05:
I habe used both, Xloader and Adurino (but version1.6).
Camera is PAL, Runcam eagle.

awolf78, 2017-04-05:
My guess is that the PAL Autodetection does not work. The camera might switch from NTSC to PAL during startup which could confuse the auto detect.
Can you please try the attached version. I have increased the delay during startup by 10 times which should make the auto detection more stable.
If that does not work I can send you a version which has PAL enforced.

[KISS_OSD_CONFIG V2.3.1 BETA.zip](https://github.com/awolf78/KISS_OSD/files/895977/KISS_OSD_CONFIG.V2.3.1.BETA.zip)

xXtrem, 2017-04-05:
Hi,

 

thank you for the fast reaction.

Unfortunately, the same problem still exists.

Additional Information:

In the short time the menu is visible, there is no response to sticks input

 

greetings

awolf78, 2017-04-05:
Then try this one, this one is forced to PAL.
If you get something like "no connection to kiss fc" - try swapping the rx and tx wires.

[KISS_OSD_CONFIG.zip](https://github.com/awolf78/KISS_OSD/files/900633/KISS_OSD_CONFIG.zip)

xXtrem, 2017-04-05:
Yes, this works. 

Greetings

awolf78, 2017-04-05:
When you are done with the config tool, let me know if this OSD version works.

[KISS_OSD.zip](https://github.com/awolf78/KISS_OSD/files/900734/KISS_OSD.zip)

xXtrem, 2017-04-05:
Very good, it works.
All kiss parameters are correct and can be changed !!
Very good is that I can now some things in the program turn on and off.
Fyi: In the config program, some icons were displayed, although icons were switched off.
In the main program, however, it is correct.

Thank you for your good work, this is very helpful.
Greetings Frank

awolf78, 2017-04-06:
First of all: Thank you for the donation!

I will check the icons in the config tool for the next release.

Question: Did you use the original KISS OSD V2.3 or the one I sent you for it to work?

Aus Deutschland?

xXtrem, 2017-04-06:
I only used your new version 2.3.1. 
The 2.3  I have not tested. I can do this today.
Yes, here is Germany.

awolf78, 2017-04-06:
V2.3 brauchst Du nicht testen, wird wahrscheinlich dasselbe Problem haben wie vorher.

Kannst Du mir bitte einen Gefallen tun und die Version im Anhang testen? Sollte die nicht funktionieren flashe bitte wieder die vorherige und gib mir kurz Bescheid.

[KISS_OSD V.2.3.1 BETA.zip](https://github.com/awolf78/KISS_OSD/files/902230/KISS_OSD.V.2.3.1.BETA.zip)

xXtrem, 2017-04-06:
okay in deutsch geht es schneller, fein.
Also, 2.3.1.Beta läuft exakt wie die alte 2.3.1 passt also.
Noch ein kleiner Bug?? Im State- Menü(nach dem Flug) wird für ESC3 fast! immer eine völlig falsche Drehzahl angegeben, selbst bei der Startdrehzahl, ohne Airmode! Z.B. irgendwas zwischen 20-50.000 statt 2000 bei den restlichen ESCs. Die weiteren Daten wie Strom, Temp sind korrekt. 
Das ist übrigens auch in V2.2 so.
Übrigens: Ich hoffe dass Kiss irgendwann mal den Frsky-RSSI durchreicht, der wäre schon nett in der Brille.
Grüße

awolf78, 2017-04-06:
Super, dann funktioniert die neue PAL/NTSC Autodetektion ;)

ESC3: Das ist leider kein bug in meinem OSD, sondern ein Problem mit Deinem ESC3. Du kannst versuchen die ESCs nochmal zu flashen, aber bei mir hat das nicht geholfen. Ich habe einen ESC der gar keine Telemetriedaten mehr liefert ansonsten allerdings einwandfrei funktioniert.

RSSI: Da brauche Ich Deine Unterstützung: Poste bitte in der KISS FC Facebook oder rc-groups das Du dieses Feature gerne hättest (am besten beide). Ich habe das auch schon angefragt, ist aber bisher auf taube Ohren gestoßen :(

Vielen Dank fürs testen, Du bist als beta Tester angeheuert :)

closed 2017-04-07 by awolf78

xXtrem, 2017-04-12:
Danke für das schnelle Feedback, ich bin erst heute wieder verfügbar.
Der KISS-Fehler mit den ESCs ist ja interessant. Vielleicht ist das ja auch die Ursache für die falsche Kapazitätsanzeige vom FC. In der Taranis muss ich dafür einen Korrekturfaktor von +30% ansetzen!
Bezüglich RSSI werde ich mit maulen, ist mir wichtig.
Beta-Tester, jederzeit ! 
Schöne Ostern

headofthemonkeys, 2017-04-13:
Hello, do you know if the OSD works with NTSC settings/cam ?

awolf78, 2017-04-13:
Yes