awolf78/KISS_OSD #18: OSD Config Tool / Move items

austrianglider, 2017-04-16:
I have the problem that I can't move the items down. 
The other move functions work (Pitch: Up, Roll: Left / Right) 
My remote control is set to Mode 1 (Horus, Kiss FC)
Each channel is set to +/-100.
I have installed the current version 2.3.1.

awolf78, 2017-04-16:
Try flashing the attached (use .hex if you use XLoader or .ino if you use Arduino).
Once flashed, connected a battery and wait 30 seconds.
Then flash the KISS OSD config tool again.

Questions:
What camera do you use?
How do you flash the OSD? With XLoader? Or Arduino?
If Arduino: What version do you use?

[eeprom_clear.zip](https://github.com/awolf78/KISS_OSD/files/924506/eeprom_clear.zip)

austrianglider, 2017-04-17:
First, I would like to thank you for your help.
Still does not work after clearing the EEPROM and reflashing the OSD config tool.
Moving the items plalced on the top of the screen works fine, also in the down directions.
The items on the bottom of the scrren like timer, esc, battery icon, nickname, ... doesn't move downwards (after moving them upwards).

Camera: RunCam Swift PAL
I use XLoader for flashing.

awolf78, 2017-04-17:
Please upload the attached version and go to "Move Items". On the left center border of the screen, two numbers should appear. The top number should say 15. The bottom number should say something less than 15. Please select the items you cannot move and let me know what numbers are reported.
Thank you for reporting this - and thank you for helping me debug this in advance.

[KISS_OSD_CONFIG TEST.zip](https://github.com/awolf78/KISS_OSD/files/925800/KISS_OSD_CONFIG.TEST.zip)

austrianglider, 2017-04-17:
**Item (top number/bottom number):**
Battery icon 15/14
Nickname 15/14
Voltage 15/14
Timer 15/13
Bottom Right ESC 15/13
Bottom Left ESC 15/13

After that i have done a reset in "Reset - Restore Defaults". Then I went again in "Move Items" and moving with ALL the items worked fine. 

Installed again the original  V2.3.1 OSD Config Tool.
Did the following steps:
.) Font update
.) Move Items : Problem with some items to move downwards
.) "Reset - Restore Defaults"
.) Move Items: Moving of all items works perfectly.

awolf78, 2017-04-17:
Thank you for the detailed explanation. I will try to reproduce this with my PAL camera. Glad you found a workaround.

pazz2576, 2017-07-16:
Hi Awolf78

ive flashed 2.4 beta on my minimosd kissfc v1.2 with 24a esc.
In display setting i can relocate the items to new positions.
I also read in the threats that it is possible to delete items of the screen.
I like to remove the esc information, its filling the screen and makes it overcrowded.
Whatever i try, this is not working for me. Is there some sort of stick combo to remove items.

tnx,

pazz2576

awolf78, 2017-07-17:
Flash the config tool again and go to DISPLAY->DV CHANNEL. It is default on AUX1, so if you did not change that and you are currently not using AUX1 on your radio then you don't have to flash the config tool. On your radio configure a knob on AUX1. Depending on the knob position you can display more or less OSD items. You can change the order they are displayed in the config tool under ORDER ITEMS.

pazz2576, 2017-07-17:
Thanks for your help. Dit not know what the dv was for. Afterwards found your wiki and folowed it. Enabled it and it works like a charm. Nice.
One other thing. Also readin your wiki about the osd menu for pid's, filters and other things. I can not open this menu. Only the recap when i disarm. Tried every stick combo to open it. Found only the stick combo for battery type. Is there some other things i could do?

Thanks

awolf78, 2017-07-17:
While disarmed, yaw left for more than 3 seconds will open menu.

awolf78, 2017-12-09:
Closing this, could never reproduce it and no one else has mentioned this

closed 2017-12-09 by awolf78