awolf78/KISS_OSD #19: OSD Config - Move Items: please add a "hide"

MichaelPallinger, 2017-05-14:
Hi,

i use mx12, which has 6 channels. all are used. it would be fantastic, if i can hide the motor infos from the borders in the move menu.

Thx,
Michi

schemen, 2017-05-14:
Hi,

I would also prefere to be able to disable certain items permanently instead of using a channel to switch it dynamically. For example, all the ESC data (single esc) is not reaaaally that interesting to me, I love a minimalistic interface =) Voltage is close to all I need, overall pulled mah is also nice, maybe current A draw, but that would be it.

Cheers!
Schemen

awolf78, 2017-05-14:
You could actually set that up using the arm switch by manipulating the minimum and maximum switch position on the radio. The arm function and the display function would be the same. In the new KISS GUI you can configure the level of the arm switch.

schemen, 2017-05-15:
Mh, so if I arm everything disappears, just the thing I need is here?

Still, I would love to see a static setting. Do you know how much work this would be?

awolf78, 2017-05-15:
If you put it on arm, you would not see any item while disarmed. Only when armed, a few items would appear - depending where you arm switch level is set at (on your radio).
The work is not the problem, however I'm a bit challenged with program space, so I have to think twice before I implement a new idea. But I thought about it yesterday and found an easy way so it would at least not inflate the KISS OSD code by much. Mostly programming in the config tool. Will take a while though, busy at work these days and won't be before the 2.4 release. So the suggestion with the arm switch is a workaround for in between.

schemen, 2017-05-15:
Okay, cool! I am currently looking into this regarding the KISS OSD PDB solution for ImpulseRC. We're looking into this as well!

Thanks for your input!

awolf78, 2017-05-22:
You can now set the "DV CHANNEL" to "FIXED" which enables you to statically turn on and off OSD items. Will be part of 2.4 release, commited in master.

closed 2017-05-22 by awolf78