awolf78/KISS_OSD #21: Voltage reading from FC instead of OSD

mkmitchell, 2017-07-08:
I use a [reactor PDB/OSD](https://droneplastics.com/Attachments/Reaktor_PDB_OSD_Install_Rev3.pdf) that I prefer to use for voltage readings.  My FC is a little messed up and only reads 5v instead of battery.

Is there any configuration to change the voltage to read from my OSD instead of the FC?

Thanks

awolf78, 2017-07-08:
I guess you are not using KISS 24RE ESCs either - otherwise it would use the voltage measurement of those instead of the (less accurate) reading from the FC.
If that is the case, you better stick with MWOSD.

mkmitchell, 2017-07-08:
Hmm. I am using the kiss 24a escs. I'm testing it today so I'll let you
know what sort of readings I'm getting.

mkmitchell, 2017-07-09:
I wasn't getting any ESC readings either.  It very well may have been the messed up fc.  I decided to just tear it all apart.

closed 2017-07-12 by awolf78