awolf78/KISS_OSD #22: issue with updating W/ xloader steele osd pdb

WestWi, 2017-08-05:
steele osd pdb with all kiss hardware

i try to upload the setup tool and have uno (A Tmeg328) selected along with the hexfile  configtool. com 2 port is unplugged.
i hit upload and it says uploading but never finishes. even left it for several minutes and never finished.
thank you for your help!

gotnull, 2017-08-05:
Make sure both interface ports are unplugged before flashing and make sure you have a battery connected and try again.

WestWi, 2017-08-05:
Both interface ports?

gotnull, 2017-08-05:
Yes

WestWi, 2017-08-05:
Like the com 1 and com 2 ports?

awolf78, 2017-08-05:
Did you also post in rc-groups? Try the driver I suggested. Helped someone else already...

WestWi, 2017-08-05:
I did. I didn't see the driver, where would I find that? Thank you for your help with this!

awolf78, 2017-08-05:
https://www.silabs.com/documents/public/software/CP210x_Windows_Drivers.zip

WestWi, 2017-08-05:
> CP210x_Windows_Drivers.zip

i tried it but still having the same issue. it is recognized by the program
just never completes the upload

awolf78, 2017-08-05:
Did you unplug the two cables that connect the KISS FC and Steele PDB?
Did you plug in a battery?
Did you select Uno ATMEGA328 in XLoader?

WestWi, 2017-08-05:
Yes to all

awolf78, 2017-08-06:
You have a different PC you can try? Or a buddy that has already flashed it?
Otherwise next step would be to contact ImpulseRC support :(

gotnull, 2017-08-06:
I had an issue during testing on my Windows machine and then when I used a
Mac it worked fine.

It must've been an issue with my drivers, which I haven't addressed yet.

WestWi, 2017-08-06:
i do have access to a mac. i don't have much experience with mac computers
though. where could i find the mac compatible software for the x loader?

gotnull, 2017-08-06:
I've used this with success:

http://paulkaplan.me/HexUploader/

WestWi, 2017-08-06:
> issuecomment-320481763
> auth/AdUQa9HYA5iOU4gDE7Pth5zO_PjrJYuXks5sVWAcgaJpZM4OuV9t

okay ive got that set up but am unable to find the correct hex files to load

WestWi, 2017-08-06:
never mind i figured that out, that was dumb on my part. i have the
hexuploader up and running and it recognizes the board but when i click on
the file it doesnt seem to do anything. the window just closes and the hex
uploader just asks for me to select another file

awolf78, 2017-08-06:
[image]

WestWi, 2017-08-06:
I did install that, then when it tried to load it said that it was "out of sync" and tried 10 times to load then failed.

awolf78, 2017-08-06:
Go in the "Other OSD folder". Go to either the config tool or the osd folder and open either the KISS OSD CONFIG TOOL or the KISS OSD.ino. Follow the "other OSD installation guide" below, but select "Arduino/Genuino Uno" instead. There is a status display on the lower part of the Arduino window, let me know what it says there.

WestWi, 2017-08-07:
I was reading on a post that the pdb comes pre flashed. I'm not getting anything for a display. Is this an indication that something beyond the end user is able to fix?

awolf78, 2017-08-07:
That's not good.
There is a green and red status led on the board. Does the red led start flashing at all?

WestWi, 2017-08-07:
No, it's a solid green

awolf78, 2017-08-07:
Then the main loop of the OSD FW is not running :(
Since you are not able to flash it either I can only assume the ATMEGA328 is dead on your PDB.
Go to the ImpulseRC product page for the PDB and double check that you have wired everything exactly the way they described. If that is the case - contact their support for a replacement. I'm sorry this happened to you :(

WestWi, 2017-08-07:
Thank you for all your efforts in helping me with this. Everything is wired correctly. I'll get ahold of impulse rc and get a replacement. Thank you again

awolf78, 2017-08-07:
Sorry this happened to you. ImpulseRC has great customer support, they will send you a replacement ASAP.
Closing this for new, please come back when you have any more questions/issues!

closed 2017-08-07 by awolf78