awolf78/KISS_OSD #23: Voltage wrong

maganzi, 2017-09-01:
Hello,

https://youtu.be/gSINJNPbjSk

voltage is wrong, if i throttle a bit, the voltage is rising up. 
If throttle 0% the voltage is 0.1V

Setup:
TBS Silk
Full KISS,
Lumenier PDB
TBS Unify Pro
Runcam RotorRiot Edition 
Piggy OSD Furious

awolf78, 2017-09-01:
The video doesn't work, make sure you have it as "Unlisted" and not "Private" - otherwise I cannot see it.
Go in the KISS GUI, plug in a lipo and check the voltage in the upper right corner in the Advanced tab. What does it say there?

maganzi, 2017-09-01:
[image]

i tried an older version of your OSD and in this version, the voltage is working perfectly fine.
The video will work now.

closed 2017-09-01 by awolf78