awolf78/KISS_OSD #24: missing mah correction

olivier0253, 2017-09-14:
Hello,

I have 2.4 version and i would like to correct the mah consuption at screen.
As told in wiki, i go to BATTERY in KISS OSD but i don't find MAH CORRECTION.
Did i make a mistake? or this setup is available somewhere else?...

How can i do to correct the mah?

Thanks for your help

Olivier

awolf78, 2017-09-14:
I've been a bit behind on releases. Just updated it, go to the release section.

olivier0253, 2017-09-15:
Thank you for your job and your rapidity.
It works fine!
In the same way, how can i change TIMER setup (AUTO, RACE, RESET) with a 2.4 version and FC? I can't modify this setup in this version.
And i see MENU is disabled in RC4.3...

closed 2017-09-15 by awolf78