awolf78/KISS_OSD #25: Screen says "No connection to Kiss FC"

Bkauf01, 2017-09-16:
After following the setup guide from the wiki [here](https://github.com/awolf78/KISS_OSD/wiki) and going through the troubleshooting section(**Including flipping TX and RX on Port 2. Which are the two pins on port 2 closest to the middle of the board**), I am still unable to see the video from my OSD.

I am using Mr. Steele's setup with the following:

* KISS FC, flashed with latest firmware (V 2.1.0)
* Mr. Steele PDB, programmed with **KB SteelePDB.V2.4.RC4.3.zip**(using Arduino 1.8.4 on mac)

Some hopefully helpful notes:

* The fc shows accelerometer, motors run, etc.
* **I am able to see "No connection to KISS FC" on my goggles**

awolf78, 2017-09-16:
What FC do you use? You said Cleanflight? This is designed for the KISS FC and requires KISS FC firmware to run on your flight controller. Which flight controller do you use?

Bkauf01, 2017-09-17:
I use the Kiss FC with flashed with v2.1.0 for kiss fc

fedorcomander, 2017-09-17:
2.1.0 ?

do u mean 1.2 ?

Bkauf01, 2017-09-17:
@fedorcomander 

Nope, on cleanflight I selected: 

1. `KISSFC` as my board
1. v2.1.0 - KISSFC - 30-07-2017 07:26 (stable)

awolf78, 2017-09-17:
That is the cleanflight version for the KISS FC board. You need to run actual KISS FC firmware for the SteelePDB. Where did you buy your KISS FC at? Did it come flashed with cleanflight? Then it is most probably a clone...

Bkauf01, 2017-09-18:
@awolf78 I purchased from Team Black Sheep, but used cleanflight configurator to configure it. If the issue is running the actual KISS FC firmware for SteelePDB, where do I get that from, what do I use to program/configure it?

Thanks for all of your help with this.

awolf78, 2017-09-18:
http://kiss.flyduino.net/firmware-v1-3rc25-public-beta-for-kiss-fccc/

Bkauf01, 2017-09-18:
@awolf78 Awesome, thank you so much for your help with this.

closed 2017-09-18 by Bkauf01