awolf78/KISS_OSD #26: Issue with voltage correction (-)

olivier0253, 2017-09-17:
Hello,

I have (now) V2.4 RC4.3 and i meet issue when i want to make voltage correction.
When i apply positive correction, it seems to be good
However, when i apply negative one, it's run at first, but at next lipo, it's show me 42V at screen! (i run 4S)
So, i go to VOLTAGE CORRECTION and the value has move to 25V (approwimately, denpending of negative value i apply on previous lipo)
This issue occurs with kiss FC and CC.

Thanks for your job!

Olivier

awolf78, 2017-09-17:
Ok, this has not been very well tested, thank you for reporting. I will look into this probably on Thursday.

awolf78, 2017-09-21:
Try this one.

[KISS_OSD.ino.standard.zip](https://github.com/awolf78/KISS_OSD/files/1321453/KISS_OSD.ino.standard.zip)

olivier0253, 2017-09-25:
Thanks
I'll test it asap, when i go back to home.
One question before install... How i should to do?
Like a basic firmware installation, with Xloader?

awolf78, 2017-09-26:
Yes

olivier0253, 2017-09-28:
I tried this file...
Installation seems to be good with Xloader, but when i plug in the lipo, i have no more OSD :-(

awolf78, 2017-09-28:
What type of OSD hardware do you use?

olivier0253, 2017-09-29:
i use piggy osd v2, with a 470uF capacitor

awolf78, 2017-09-29:
[KISS_OSD.ino.eightanaloginputs.zip](https://github.com/awolf78/KISS_OSD/files/1344128/KISS_OSD.ino.eightanaloginputs.zip)

closed 2017-09-29 by awolf78