awolf78/KISS_OSD #32: Runcam split issue with osd connected

olivier0253, 2017-10-02:
Hello,

I come back with another config, and another issue...
With Kiss FC (1.3 RC30), i want to connect a runcam split (latest fw, 1.1.0) on tx pad. This part run fine, i can start and stop the recording via my radio.

_But_, when i connect osd (tx/rx) and split (tx) on FC, the OSD runs fine while the split does not answer.

Thanks

Oliv

awolf78, 2017-10-03:
Very strange it worked for me. Not sure if there is anything we can do about it. You can ask fedorcomander in the KISS FC thread.

olivier0253, 2017-10-03:
i asked to FedorComander (https://www.rcgroups.com/forums/showthread.php?2555204-KISS-FC-Flyduino-32bit-Flightcontroller/page1670)
So, it works for you... with the same config? (hardware & firmware?)

awolf78, 2017-10-03:
I tested mine on a quad with a SteelePDB. Not sure why it would not work in your configuration though.

dayue1984, 2017-12-10:
I have the same problem. The kiss fc can not control the rc split when kiss osd connected. My setup:
KISS CC ,1.3rc30.
TBS Unify race edition, smartaudio connnect to FC RX3.
MiniOSD,  with KISS OSD v2.4-2.5, Rc Split option is on, 5v powered by TBS unify.
Runcam Split v1.10, split RX to FC TX,  5v powered by FC.
***
And the kiss osd will be blacked or freezed when the split's autos-record option is on, the split stop to work after the FC inited.
I have to turn off the auto-record option, and tried to wire all GND together, but the FC stll can not control the split.
Of cause, the split works fine with my KISS FC without osd connected.
It seems that the osd has some conflicts with the split while sharing the same serial port.

awolf78, 2017-12-10:
I have updated the release section, download KISS OSD V2.5 again, flash the config tool and turn the RC Split on under MISC. Connect the RC Split control wire (RX on RC Split) to the current pad on the minimOSD. Then it will start/stop recording when you arm/disarm.

closed 2017-12-13 by awolf78