awolf78/KISS_OSD #34: voltage mah and amp draw stop working after first battery

loosepropfpv, 2017-10-30:
ok guys so after finish building the mr steele alien everything works great on the first flight voltage was working perfectly.(no crash) so to the second battery mah and amp start reading 0 and half way though the pack voltage was stuck on 15.6v so i came back put new battery in and now it only read the starting voltage can any one please help??

awolf78, 2017-10-30:
What about the mAh on your Taranis? Is that 0 as well?

loosepropfpv, 2017-10-30:
hi no is reading out correctly

loosepropfpv, 2017-10-30:
and thank you for getting back to me so quick

awolf78, 2017-10-30:
When the voltage is "stuck" in the OSD: can you Yaw to the left while disarmed for a few seconds to see if the menu still pops up?

loosepropfpv, 2017-10-30:
yes the menu still popped up

awolf78, 2017-10-30:
Very strange. Since your Taranis is reporting mAh correctly I am assuming the ESC telemetry is still working. And since you can get in the menu in the OSD, the communication between FC and OSD works properly. Somehow the FC is not reporting the ESC telemetry correctly for some reason - I would try flashing the FC first and then the PDB.

loosepropfpv, 2017-10-30:
hi there i just accidentally fix the problem. when i was in the telementry screen i press the wrong button and it delete all the data and when i press all the discover new sensor all my bam all osd is now reading correctly!!! thank you for your time!!! but is all fix!!

loosepropfpv, 2017-10-30:
i will leave this thread open till test flight tomorrow

loosepropfpv, 2017-10-30:
thank you for your work in the fpv community and hope you carry on on what you doing!!!!

awolf78, 2017-10-30:
So you fixed it by doing a rediscover on your Taranis?

loosepropfpv, 2017-10-30:
yes by accident!!!!

awolf78, 2017-10-30:
Interesting, who would have though that has anything to do with that...

loosepropfpv, 2017-10-30:
maybe it sorted itself out????

loosepropfpv, 2017-11-01:
Hi again so now I have the same issue however this time sometime it works and sometime it don’t for example in mid flight it will work for 30sec then it will freeze all amp mah and voltage stay the same and then it will work again

awolf78, 2017-11-01:
Try flashing your KISS FC again. I don't think it is the PDB since it does not report any connection loss from the FC. The FC just keeps reporting the same telemetry data intermittently, so something is wrong on the FC.

loosepropfpv, 2017-11-01:
Perfect my friend thank you

loosepropfpv, 2017-11-01:
Hi again I have now reflash the fc and osd work for 4sec now is saying no connection to fc

loosepropfpv, 2017-11-01:
Restart it now and gain osd

awolf78, 2017-11-01:
So it is working now?

loosepropfpv, 2017-11-01:
Hi again let start this again I found now that the telematary is not working right on my taranis aswell the freezes aswell so I don’t think is the osd

closed 2017-11-04 by awolf78