awolf78/KISS_OSD #35: Crossfire rssi in osd

CraigFPV42, 2017-11-17:
Hi, 
I have set up the osd according to this link here:
https://github.com/awolf78/KISS_OSD/wiki
I have followed the instructions the only difference is that I have put it to Aux 4 in the taranis following the link where Oscar describes how to do it for frsky. 

When I plug in the kiss fc to the computer I can see the rqly signal in the aux 4 signal so I know its working correctly but it jumps between 1300 and 1995 when off/on. In the osd configurator I put those numbers in and set it to aux 4, but don't get anything other than 0 db in the osd. 

Also I've noticed that Arduino says the firmware is very large for the chipset it takes like 98% of it. Any clues as to what is going on here. I purchased this one: https://www.flyduino.net/en_US/shop/product/pr1865-micro-minimosd-kissfc-osd-fw-preflashed-pal-2683?search=Osd

Unfortunately it didn't come with the newest firmware on it and it's been a pain updating it.

awolf78, 2017-11-20:
I will double check this weekend and finally get it fixed. Thanks for your patience.

CraigFPV42, 2017-11-20:
Thank you. Is it normal that in Arduino it says there is barely enough space? Is there a way to wipe it clean to free up some space?

awolf78, 2017-11-20:
That is normal and has nothing to do with the RSSI issue.

Drewlrox, 2017-11-21:
With micro minimosd on v2.4RC4.3 I needed it set from 0-1000 but with steele pdb on same firmware it works on -1 and -1. Not sure why but this worked for me.

awolf78, 2017-12-03:
[Steele PDB v2.5 beta OSD.zip](https://github.com/awolf78/KISS_OSD/files/1525241/Steele.PDB.v2.5.beta.OSD.zip)
[Steele PDB v2.5 beta CONFIG.zip](https://github.com/awolf78/KISS_OSD/files/1525242/Steele.PDB.v2.5.beta.CONFIG.zip)

Try this one. Go in the config tool first and setup min and max RSSI under DISPLAY->RSSI.

awolf78, 2017-12-09:
Fixed with V2.5 release

closed 2017-12-09 by awolf78