awolf78/KISS_OSD #38: OSD doesn't work with kiss V2 F7

S-T-A, 2018-02-01:
I uploaded the configuration FW it worked but the OSD fiction to access the pid and vtx doesn't work off the sticks with the Steel's pdb OSD and the latest kiss V2 FC when Tring to enter the set up screen!

awolf78, 2018-02-01:
You need to yaw left and hold it for more than 2 seconds for the menu to pop up. Quad must be disarmed of course.

S-T-A, 2018-02-02:
Ok you have a valid point but my if my FC when I plug it into the kiss GUI
it tells me it's disarmed and as soon as I plug in a battery it says armed
when it's not I know for a fact it's not so what would the next step?

S-T-A, 2018-02-07:
is there a problem with the kiss V2 FC and and your steels OSD when it says
armed when it is in fact disarmed?

S-T-A, 2018-03-04:
Ok on v2.5 and kiss v2 installing OSD configuration works. I used my transmitter it worked fine, saved the configuration. I then loaded the OSD  that took plugged back in the port 2. When it starts up it shows the FC is armed when the arm switch isn't In arm position and says FC connection??? I checked all the wires have continuity I even tried switcing the tx and Rx wires and nothing I'm stumped I don't get it? It should of worked since I could program it just before I flashed the OSD hex.

awolf78, 2018-03-04:
Unplug both ports from the PDB and try flashing it again.

S-T-A, 2018-03-05:
I tried 4 times there is either something in compatible on the v2 or the
pdb firmware. I think there is an issue it gets input from FC to do the
configuration that works no problem just when you go to load the OSD
firmware on to it it would say no connection to FC or nothing. I'm swamped
the port 1 and 2 wires with each other to eliminate the wires being a
problem and the their not it arms and flys the OSD works! But when you
first plug in it after impulse logo it says armed!!!! Now in the FC, I have
armed on aux 1. Min command is 1075 I tried making it 1000 no difference I
can't get it to start up with out saying arm so I can never get into the
OSD firmware to make any changes is there a problem with the kiss v2 or the
pdb??? I have another v2 with 32amp kiss esc's and a Steel's pdb which I
was going to use in a reverb. I'm extremely frustrated and would really
like to get it working should I buy a older kiss 1.03 v1? I'm pretty sure
is should work with an older one no???

awolf78, 2018-03-05:
The SteelePDB does work with the V2, there are many users that are running that combination. Did you check in the KISS GUI if you have ESC telemetry? Connect USB to FC, run KISS GUI, connect a Lipo (props off) and go in the advanced tab. On the drop down you can display the various ESC telemetry items such as mah, rpm, etc. Please double check that is working properly.

S-T-A, 2018-03-06:
Just checked in the GUI the telemetry is not working in the final OSD
firmware but when it's the test I can set it up and every thing else I
switched the cables and checked them for continuity and tried switching the
tx and Rx and nothing so is it the kiss v2 maybe I have everything selected
correcty? I just don't get how come I can flash the configuration file it
works and responds to my radio and does what it's so pose to do when I
flash the final OSD bam I get kicked in the nuts! :(

awolf78, 2018-03-06:
If you do not have any telemetry in the KISS GUI, then you might have a short somewhere on your ESC telemetry. Disconnect the telemetry wire for each ESC one by one and keep checking in the KISS GUI if you get ESC telemetry for the ESCs that are still connected. That might help to narrow down the culprit.

S-T-A, 2018-03-06:
Aren't the telemetry wires all a common signal going to the OSD then the
fc?

In beta flight4f with beta flight esc's on my reverb the telemetry wires on
the pdb all go to a common wire exiting the pdb going to the fc.

awolf78, 2018-03-06:
No, the telemetry wires are connected to the FC. FC reads ESC telemetry. OSD reads telemetry (stick input, etc.) from FC.

S-T-A, 2018-03-07:
Do you have a electrical schematic of the OSD pdb?

awolf78, 2018-03-07:
I don't. You can ask ImpulseRC, but I highly doubt they will give you the board design showing all the traces. All you should need is the pictures on their website:
https://impulserc.com/collections/pdbs/products/mr-steele-alien-pdb-kit-for-kiss-with-osd-and-mic

S-T-A, 2018-03-25:
ok this is my second PDB and it's exactly the same as the first i flashed
and connected my KISS V2 and steels PDB in my reverb and kiss 32A esc's no
shorts nothing of the sort. When the KISS OSD is flashed when first pluging
in it says armed!!! when it's not and the Kiss GUI 2.0.6 says its disarmed
and serial is set to OSD! and there is no telemetry feed back through the
PDB To The FC in the GUI. yet it arms and responds just can't access the
main menu and the stick doesn't respond the final firm ware flash...

So what the next step???????

awolf78, 2018-03-25:
Try changing arm to AUX1 or AUX2 - that has helped some other users

S-T-A, 2018-03-25:
I will try that thanks wolf 😎

Alephnull27, 2018-03-27:
Hi, same problem here. Kiss V2, and mr Steel PDB. Arming on AUX1 or AUX2 didnt work for me. Also dont have telemetry for ESC on KISS GUI, but i can see telemetry of ESC on OSD. 
Same as S-T-A, i cant access the OSD menu by holding yaw on left. 
I think there is some kind of incompatibility. Hope you can find and solve it quickly :-)
Meanwhile... any other ideas or workarounds?

awolf78, 2018-03-27:
Disable level mode

Alephnull27, 2018-04-05:
thanks! worked for me!

S-T-A, 2018-04-05:
I set up arm from aux 1 to 2 and it worked but either aux 1 or 2 when
moving the switches it showed the disarmed/armed only when I changed the
armed from aux 1 to 2.

But the ESC telemetry doesn't work in the kiss GUI on two of my Mr. Steel's
pdb OSD???  Could that be changed in it the next firmware update?

 Also would it be possible to change the size of the font so if we wanted
the ESC telemetry it would be smaller then the rest not to take up the
whole screen I was thinking 1/3 of the current size? So we could put it on
the side of the screen even when I do it's still too big.
Kick ass OSD wolf!!!

awolf78, 2018-04-07:
If the ESC telemetry does not work in the KISS GUI you most probably have a short on one of the ESC telemetry wires (they are internally connected together on the KISS FC). As I suggested before, try disconnecting the telemetry wires one by one to find the issue.

I would recommend you to use a switch or better a knob on your radio to control how many OSD items are displayed - especially useful for the ESC telemetry which you do not need to see all the time. In my wiki check out the section "Making your OSD more or less busy", link is below.
https://github.com/awolf78/KISS_OSD/wiki#setup-of-kiss-osd

S-T-A, 2018-04-07:
Two separate builds and trust me when I say there are no shorts 100% quite
simplely it doesn't work! On the kiss v2 kiss 32amp ESC

awolf78, 2018-04-07:
Did you check the pins on the KISS FC and PDB? If those are not bent either, then I would talk to your dealer about it. It has nothing to do with the OSD FW - if the FC does not receive telemetry from the ESCs it cannot send it to the OSD and the OSD cannot display it. And since you cannot see the ESC telemetry in the KISS GUI the issue is on the ESC<>FC side.

S-T-A, 2018-04-07:
There not bent bit 2 different kiss v2 FC's and ft so different Mr. Steel's
pdb's? Really odd?

closed 2018-09-27 by awolf78