awolf78/KISS_OSD #39: Strange Icon

danricho, 2018-03-09:
[image]

I'm encountering a strange issue where the minimOSD is adding a strange character as shown in the attached image.

It happens after the font has been loaded and persistents on both the config and standard hex firmware. It doesn't change position or disappear when the menus are accessed either.

Any ideas?

awolf78, 2018-03-09:
The character on the top right is the Wattmeter - you can turn that off in the config tool under DISPLAY->WATTMETER.
The character in the middle of the screen should not be there at all. It is part of the alternative Wattmeter (beermug). I am suspecting that the font memory and/or your OSD chip on your micro minimosd might be defective.
Try turning the Wattmeter off first - if that does not help you will have to get new micro minimOSD. At least those are not very expensive.

danricho, 2018-03-11:
Understood. I was mainly talking about the centre icon.

I have ordered a new minimosd and will report back.

Thanks for the assist.

closed 2018-04-21 by awolf78