awolf78/KISS_OSD #42: ESC Telemetry Not Showing

mattjsmithone, 2018-04-21:
Hi,
Has this issue with the stock OSD so decided to give this one a go as it seems to be a lot more functional!
When the quad is armed and the motors spinning, the RPM on the OSD does not change, and neither does the Current or mAH.
After the quad is disarmed and the statistics are opened, none of the values read anything but the defaults. (0A, 32F, 0RPM, 0V etc.)
All of the ESCs and the FC are updated to the latest versions, and the TLM connections are soldered

Thanks in advance!

Setup: KISS V1 FC, 24A KISS ESCs, Mr. Steele PDB.

awolf78, 2018-04-21:
The ESC telemetry is not properly received by the FC, that's why you cannot see it in the OSD. Check if you do not have a short on any of the ESC telemetry wires. All 4 ESC wires connect internally to one UART on the KISS FC (through the SteelePDB). Check if none of the pins on the SteelePDB and KISS FC are bent.

mattjsmithone, 2018-04-21:
Hi!
Thanks for the insanely fast reply!!
Just disassembled the quad and not found any bent pins and can not see any shorted wires.
Any other Ideas before I begin to unsolder the TLM wires one by one?

awolf78, 2018-04-22:
Nope, that is unfortunately what you must do at this point. Check the ESC telemetry in the KISS GUI after unsoldering each ESC telemetry wire.

mattjsmithone, 2018-04-22:
And to see the telemetry in the GUI you have to plug a battery in and go to the ESC flasher tab yes?

mattjsmithone, 2018-04-22:
Seemed to solve the problem,
re-soldered both ends of each TLM wire on the ESCs and PDB, this meant that I had to unplug the VTX.
Then I looked at the data output tab in the GUI and the temperatures were shown, so I began to put it back together. 
I tested it after putting it back together and one ESC didn't light up and the rest did not show up in the data outputs. So after a lot of messing around (3-4 painful hours), it turned out to be the VTX that was causing the issue. I had soldered the smart audio to the TX tab and not the RX tab. After I sorted that out it began to work!

Thanks for all of the help, Loving the OSD now!

TL;DR:  Smart Audio goes to RX not TX on the FC and can cause problems with the TLM

closed 2018-04-22 by awolf78