awolf78/KISS_OSD #43: No spinning props with Icons ON

ryusko2, 2018-04-26:
Setup:
KISSFCV2F7-1.3RC32e
Kiss 32A 32-bit ESCs, full telemetry
KISS_OSD 2.5
Micro MinimOSD+470uF cap

I am unable to get spinning prop icons only - I now realize this could be due to the telemetry I have hooked up to the FC (if OSD detects rotation telemetry from ESC, I am guessing it displays this information instead of spinning prop icon); however, it would be nice to see only the spinning props, as I prefer this display to the rotation telemetry.  Or even better, if the props could be an additional item in the order, so we could turn the spinning icons on/off based on our set order...

awolf78, 2018-05-05:
If you change the order (in the config tool) so the ESC RPM is first and the ESC amps and voltage is afterwards it should do what you are looking for. FC must have ESC telemetry though, otherwise OSD cannot pull ESC telemetry from FC.

ryusko2, 2018-05-07:
This is my order:

- Voltage
- Timer
- Throttle
- Callsign
- ESC RPM
- RSSI
- MAH
- WattMeter
- ESC Amps
- ESC Temp

It displays the spinning props properly as long as KISS_OSD_CONFIG is loaded, but as soon as KISS_OSD is loaded, the spinning prop icons change to a '0.0 KR' followed by ESC icon, then the numbers update based on (kilo) rotations from telemetry ... sans spinning props icon.

ryusko2, 2018-06-06:
Sorry for the delay, Alex...Memorial Day got the better of me.  Your attachment works as expected now :)  I've just sent you some dough for the fantastic work!  Unfortunately, I'm just not able to get Kiss v2 + 32A ESCs to play nicely together (the ESCs are drawing 15% more power, and I can't seem to tune my PIDs to satisfaction, even though I've tuned 10+ in betaflight no problem), so my build will sit on the table for awhile :(

k1berj01nt, 2018-09-03:
Is this one taken into account in fresh release?
Used to flash this version last time, as had same issue described above..
Best regards, and many thanks for the OSD!!!

aQuaplug, 2023-07-14:
> This is my order:
>
> * Voltage
> * Timer
> * Throttle
> * Callsign
> * ESC RPM
> * RSSI
> * MAH
> * WattMeter
> * ESC Amps
> * ESC Temp
>
> It displays the spinning props properly as long as KISS_OSD_CONFIG is loaded, but [...]

Exactly same issue with Steele PDB v1 for Alien. I'm using KISS V1.
And some other icon and characters also changed when flashed SteelePDB V2.5.1 OSD.
Throtle changed % to %T, DBI changed to % and so on.