awolf78/KISS_OSD #44: COM3 doesnt show up in the tools ports tab

timmyboi57, 2018-04-29:
COM3 does not show up in the tools>ports tab tried to upload using a different port: /dev/cu.Bluetooth-Incoming-Port says not in sync

awolf78, 2018-05-05:
Sorry, not much of a MAC pro here. From what I heard it should work out of the box with a MAC (and not too ancient OSX).

closed 2018-05-28 by awolf78

fedorcomander, 2018-05-28:
Port name you are trying to use says BLUETOOTH!

/dev/cu.Bluetooth-Incoming-Port

Pay attention. On mac works out of the box, or check CP2102 drivers from Sylabs.

https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers <https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers>

Also, there was some issue in previous mac osx, but fixed in high sierra. 

/FedorCommandrr