awolf78/KISS_OSD #46: Can't set RSSI channel to AUX7

mReXiTuS, 2018-05-25:
Hi,

This is not issue, but I would really appreciate if you can add support for RSSI channel up to AUX8 (channel 12). I'm using FlySky i6X radio which can provide RSSI only on channel 11 (AUX7). Since It cannot be changed in radio, It would be super cool if I could set RSSI channel to AUX7.

Thank you,
Dominik

awolf78, 2018-05-25:
What hardware are you using? micro minimOSD? SteelePDB?

mReXiTuS, 2018-05-25:
It's micro minimOSD but kiss version: https://kiss.flyduino.net/dwkb/kiss-osd/

mReXiTuS, 2018-05-27:
Hi Alex,
I tried to do it myself, but I'm struggling with loading values from FC for AUX5-8. Would you be so kind and find 5 mins to check what am I missing? 
https://github.com/mReXiTuS/KISS_OSD/commit/3b9993cda94250239b476f6346d68fe2fc59a8a5
Thanks,
Dominik

awolf78, 2018-05-27:
You also need to adapt the config tool.
I would have done it, but have no time currently due to heavy travel schedule work wise. Feel free to make PR if you want ;)

mReXiTuS, 2018-05-27:
I adopted both config tool and OSD, but problem is that it doesn't work yet :( I think I have problem in protocol part, since it's not properly loading aux channel values from kiss FC. I simply extended loop in ReadTelemetry method but it's apparently not enough. I noticed there is also something like bf32 mode, what's that?

awolf78, 2018-05-27:
Look in:
https://github.com/flyduino/kissfc-chrome-gui/blob/83eb07baecfb906da1ac7fe83420eae4b9356293/js/protocol.js#L314

Felix added the extra AUX channels at the end of the telemetry byte stream. Make a second for loop with the offsets for the extra AUX channels 5-7.

divenet, 2018-07-02:
just a quick question, on the same topic, aux 5 to 7 is it fixed or on going in both configurator and OSD firmware?

awolf78, 2018-07-03:
AUX 5 to 7 cannot be selected in the OSD (or config tool). I actually tried to add them, but ran out of program space.

divenet, 2018-07-04:
@awolf78  Thank you so much for your time, i should of gave a second thought and just move my channel around on FC V2 to fit your OSD firmware, i feel simple right about now. All sorted, thanks again for a great OSD and awesome work

PS any OSD Firmware updates in the pipeline or this one is set in stone for now

awolf78, 2018-07-04:
When new KISS FW comes out I will make a V2.5.1 release for D setpoint support. But otherwise you will not find any big changes here since I ran out of program space.
I have been working with ImpulseRC to develop a new OSD with a 32bit MCU (F3) which will be released next week the latest. It is based on this FW, but has way more features and is much easier to flash and configure.

divenet, 2018-07-20:
@awolf78 thank you so much for your time and input in the minim osd, I’ll be looking forward to get the new firmware 2.5.1 when it comes out and use it in my quad set up for it. I did however just got delivery of the impulse rc wolf pub and osd, I most admit you did an awesome job on that, loving every bit of it. Great work my friend

awolf78, 2018-07-20:
Glad to hear you like it!
And thanks for reminding me that I need to make a new release ;)