awolf78/KISS_OSD #47: Polulu FTDI drivers installation not working

KwadUpFPV, 2018-05-28:
Hi,
This is my second build with Steele PDB, Kiss FC V2 & Kiss 32A ESCs.
First one worked flawlessly.
Now i'm having a constant red blinking led on the Steele PDB and i don't know why.
I have ignored the blinking led now and just proceeded with flashing the Kiss FC V2 and the Kiss 32A ESCs and that worked just fine.
But now i still have that red blinking led (next to the continuous green led) and i still don't know what it's about.... 
I then decided maybe i should try and RE-flash the Steele PDB?
So as i don't have the initial Windows PC anymore (moved to Mac) i installed bootcamp on MacOS and the latest version of Windows 10.
When using Xloader, the COM port is not showing.  So it's impossible to upload a hex file... 
I have now tried to install the FTDI drivers (Polulu) but i get an error messages everytime saying:
"The current language is not supported by the Device Driver Installation Wizzard."
I'm running out of ideas...  Anything else you have in mind?  Any hints or tips?
Any advice would be much appreciated.  Thank you.

awolf78, 2018-05-28:
The red blinking LED is the "heartbeat" of the OSD. That means the OSD program loop is running. So nothing to worry about.

closed 2018-05-28 by awolf78

zitemedia, 2018-05-28:
Hi, ok, that sounds like good news then :).   But then why do I NOT have a blinking red led on the Steele PDB of my first build from some months ago?  :-/

awolf78, 2018-05-28:
I'm guessing the OSD is wordking fine on the other SteelePDB that is not blinking: then the LED is broken.