awolf78/KISS_OSD #49: AIKON 4 in 1 current sensor is not working on micro minim OSD

Heidi87, 2018-07-05:
Hi,

I would like to know if the current sensor pin on the micro minim OSD is actually used within Kiss OSD 2.5.
My setup: AIKON 4 in1 ESC on Kiss v2 works great with the micro OSD, except of the current sensor.
Since the AIKON dose not support ESC current telemetry, I tried to connect the single current sensor to the micro OSD current pin, but it did not work. Can you please help me to solve the issue?
Thank you very much in advance!
Jan

awolf78, 2018-07-05:
Sorry, but analog current sensor is not supported by KISS OSD. ESC telemetry is one of the core features for KISS, so my OSD relies and takes advantage of it as much as possible. Adding current sensor support is also not possible since I ran out of program space.
You are better off using MWOSD instead if you use a current sensor.

Heidi87, 2018-07-05:
Ok I understand. I read about your new OSD with Impulsrc. Dose it support analog current sensor and will the size of it be similar to the micro minim osd? 

Thank you very much for the prompt answer it’s really helpful :) !!!

awolf78, 2018-07-05:
The new OSD is on a PDB (called WolfPDB). So that won't be the greatest fit with a 4in1.

Heidi87, 2018-07-05:
Ok sound’s like no option for me. But thank you for the support anyways!

closed 2018-07-05 by Heidi87