awolf78/KISS_OSD #50: No Video Feed No OSD

TheFezzter, 2018-07-24:
Hello, 

I'm having issues just getting a video feed into my fatshark goggles. I can't seem to get any kind of video feed or even OSD to appear. I have flashed both the latest Steele SteelePDB.V2.5.1.zip and SteelePDB.V2.4.RC4.3.zip in hopes that maybe the older firmware would allow something to function and still nothing. 

I have test flown the drone by line of sight and the drone responds and arms to my controller inputs but its the video feed that i can't seem to get to work. Below is a link to a photo of my soldering points and to what devices i'm using for VTX and FPV Camera.  

I have those listed as well:

RunCam Swift 2 Rotor Riot D-WDR 600TVL 2.5mm Lens OSD MIC DC 5-36V CCD

TBS UNIFY PRO V3 5G8

Kiss FCv1.03 Firmware: RC34c

At this point i'm going to see if soldering another VTX would work but when plugged in all LEDs lightup on both the PDB, FC, and vTX so i'm just not sure why i wouldn't get a video feed. 

[https://imgur.com/a/TjZpam1](url)

awolf78, 2018-07-25:
You can try soldering your VTX directly to the camera - unsolder camera and vtx from PDB and connect them directly. If you still get no video feed, it is not the PDB or OSD chip.

awolf78, 2018-07-25:
Any FPV camera works with the SteelePDB.
Can you measure the voltage on your camera and VTX pads and check if they seem correct?

TheFezzter, 2018-07-25:
At the moment, i can't i don't have a voltage meter, i'll need to purchase one to check that.

closed 2018-09-27 by awolf78