awolf78/KISS_OSD #51: Steele PDB - No connection to FC

ULTlMATE, 2018-08-03:
Hi,
I am new to KISS and I have quite a hardware problem with Steele PDB. I was flying a bit with my new quad (I have 7 batteries flown on it) and after changing battery, I ran into No connection to FC problem, which seemed like there was no power to FC. When i got home i read about this and checked if any of connection wires are broken or if any of pins are bent, but there was no problem with that. So i tried connecting FC to PC which had no problem (It was powered by USB and there was no problem to connecting it to KISS GUI). So I went back on PDB and discovered that there is no connection between any of 5V outputs and PORT pin and there was same issue with V_out. So there is probably something wrong on PDB. 

The questions are, is there some option to get that working again ? Should I just solder Vbat to FC let it be (if it is safe) or should i claim PDB as faulty and send it back to seller ?

Setup: 
Mr Steele PDB V2.5.1
KISS FC v2 v1.3RC34c 
KISS 32A regulators

Thanks for any help,

awolf78, 2018-08-03:
The 5V is not connected/powered from the PDB, but the fact that you do not have any power on V_out is not good.
Connecting an extra wire from VBAT to + on KISS FC should fix this.

ULTlMATE, 2018-08-03:
Thank you very much Wolf :)
Even I am not really big fan of this solution, it works as new. I just hope nothing wrong will happen, since the board is for sure damaged in some way.

closed 2018-09-27 by awolf78