awolf78/KISS_OSD #52: Req: Digital font.

mixMugz, 2018-08-06:
Hi, i'm creator of Digital font (betaflight).

And i want ask one question. 
It this possible to add these font in KISS OSD?

https://www.dropbox.com/s/uk1u8qzswbc973r/f8.jpg?dl=0

awolf78, 2018-08-06:
KISS OSD is not setup for different fonts - BUT: ImpulseOSD which runs on the WolfPDB is:
https://impulserc.com/collections/pdbs/products/wolf-reverb-pdb-osd-kit

We would be delighted to include the font in a future release of the ImpulseOSD GUI. The GUI enables you to flash various fonts.
Please note that ImpulseOSD is a closed source project, so we need written permission from you.

mixMugz, 2018-08-07:
Yes. I already gave permission for WolfPDB. Same about the whole ImpulseRC projects.
If you need property font - i will give you MCM file. Just give me template.

awolf78, 2018-08-07:
Great, thanks for giving permission for ImpulseRC projects!

It won't fit in this firmware unfortunately. I ran out of space, hence why the WolfPDB was created in a cooperation between ImpulseRC and me.

The good news is that ImpulseOSD will be the future, and KISS OSD the past. All newer OSD hardware made by ImpulseRC will use ImpulseOSD and use your font!

mixMugz, 2018-08-07:
Yes. It's great. But i really love Mr. Steele PDB with OSD (Alien frame still good) and glad if it will work with my fonts ;)

mixMugz, 2018-08-07:
Any chance to made custom firmware for Mr.Steele PDB for Alien with only my font? (Just for me lol =)

awolf78, 2018-08-07:
The mcm file for the font is in my repository. You can modify the mcm file with an editor of your choice and then post it here. I can then compile you a customized version of the CONFIG TOOL. You need to flash that and update the font to get your font.

mixMugz, 2018-08-31:
Hi! There is a font file. Check please, if it will be ok - use it ;)
https://www.dropbox.com/s/b4q5w42myj2d5f5/kisspdb.mcm?dl=0

awolf78, 2018-09-06:
[SteelePDB V2.5.1 CONFIG TOOL special font.zip](https://github.com/awolf78/KISS_OSD/files/2358712/SteelePDB.V2.5.1.CONFIG.TOOL.special.font.zip)

Flash the attached config tool and the update the font. Let me know if everything worked correctly, then I can add it to the releases. Thank you very much for your contribution!

mixMugz, 2018-09-07:
Oh. Thanks. But now i cant check, because my quad not assembled. Can you check please? =) Thanks

awolf78, 2018-09-07:
I don't have SteelePDB setup ready either. Let me know if it worked when you have your quad back together!