awolf78/KISS_OSD #53: KISS OSD and new KISS V2 Firmware RC36i Camera Control Issue

BikeGuy1992, 2018-09-26:
I've tried to replicate this, but the new firmware from Flyduino does not allow access to the camera control function to work. 

The video from Benjamin from Flyduino shows it does works. Take note, there is no KISS OSD hooked up, the OSD shown is the stock OSD that comes with the newer Runcam cameras. 
https://www.youtube.com/watch?v=KtFwp2Ue-y8&feature=youtu.be

But in my case I have a Furious FPV Piggy OSD V2 hooked up, and it doesn't work. I'm pretty sure this would happen with a Steele PDB as well
https://www.youtube.com/watch?v=KFWMEGbBB_U

Awolf78 said this on the Flyduino Official page when I brought the issue up. 
-That should not interfere with each other. The OSD and the FC operate independent from each other - the OSD just pulls the stick positions from the FC all the time. While it may be a bit annoying that the battery menu pops up - you are not going to hold yaw to the right for too long.

This is not true. The OSD does interfere with the camera control, and you cannot access the camera settings with a OSD hooked up. Why does the OSD share the same stick command as the Runcam control device? Can this not be changed in the code to bottom yaw right for the OSD battery info and center yaw right to activate the camera control function? 

If this can be looked at, I would appreciate it a lot. I upgraded the FC firmware to RC36i to take advantage of this function on my racing quads. Sometimes we have to change the camera settings on the fly, and I was hopping to have this working especially when flying outside since the lighting changes a lot on us from cloudy to sun breaks, etc.

BikeGuy1992, 2018-09-26:
Thanks for the reply Alex.

What I can tell you is I have the receiver soldered to the default receiver
pads (serial 2 on the V2 FC), Piggy OSD is plugged into port 1 (serial 1 on
V2 FC), the KISS 4in1 ESC communication plug to PDB connection, and the
camera control for the Runcam is plugged into port 3 (serial 3 on V2 FC).

I did contact Flyduino, but they haven’t gotten back to me yet on what’s
going on after I made the video I have in the Github page.
Either the four FCs I tried it on are bad, or like you said yourself
Flyduino doesn’t have something right in the firmware that is causing this
issue.

In the video Benjamin sent me, they show no KISS OSD hooked up. The OSD
present is the built in camera OSD.

I’ll try it your way first when I get home from work with no OSD hooked up
to verify.
Thanks again for the help!

Zaine (BikeGuy)

BikeGuy1992, 2018-10-09:
Good news, discovered this completely by accident but I figured out how to
get the camera control adapter to work.

So I have double arming, the same as Mr Steele does on his quads. Turns out
my secondary switch to activate the gyro needs to be 'on' for the camera
control to work.

Once flipped the blue light does come on and I can access the camera OSD
menu. But this brings up a new issue.

The KISS OSD stays on when you're in the camera's OSD menu. Could you
somehow add a feature in the KISS OSD menu so you could 'hide it' once
you've selected it and save+exit the menu but then still yaw left to get
back into it once its off? The OSD would not be visible and in the way due
to the overlay it displays on the camera feed when it is active. Once done,
then the user can turn the KISS OSD back on after yawing left to go back to
the KISS OSD, select the ' hide OSD' to off, save+exit after they're done
making changes to the camera OSD menu.

That's the only thing I can think of to make it easier to do camera setting
adjustments for someone who has A LOT of info displaying on their OSD. Just
a thought.

Thanks!

Zaine (BikeGuy)

On Wed, Sep 26, 2018, 4:07 PM BikeGuy <zstapleton@gmail.com> wrote:

> Thanks for the reply Alex.
>
> What I can tell you is I have the receiver soldered to the default
> receiver pads (serial 2 on the V2 FC), Piggy OSD is plugged into port 1
> (serial 1 on V2 FC), the KISS [...]

awolf78, 2018-10-09:
Well the gyro does not need to be on for cam control - but that is a KISS FW issue.

You could setup REDUCE ITEMS by flashing the config tool and assign a switch on your radio that would hide all the OSD items. You need to play around with the endpoints of the switch so you only get the OSD items you want. More info here:
https://github.com/awolf78/KISS_OSD/wiki#setup-of-kiss-osd

BikeGuy1992, 2018-10-09:
Okay that's good to know. But wouldn't it be easier to just add a ' hide
OSD' feature in the menu? Way less of a hassle than going through all the
hoops of assigning and getting a knob to work in my opinion.

Zaine

awolf78, 2018-10-09:
Yeah for sure. I have 6 bytes left in the firmware. If you manage to squeeze this feature in 6 bytes - I can add it ;)

BikeGuy1992, 2018-10-09:
Even if that feature couldn't be added, is there a video or something out
there showing how to setup a knob so you could 'hide' the OSD?

I'm just thinking for the person who has no idea what their doing and this
is their first time setting sknethibg like that up. Keeping with the keep
it super simple theme 😊

Zaine

awolf78, 2018-10-09:
https://www.youtube.com/watch?v=fGuI5hb06yE&t=1s