awolf78/KISS_OSD #55: No connection to Kiss FC w/ steele pdb (Kiss FC V1)

Heligrom, 2018-10-05:
Setup:

Kiss FC V1 (latest firmware)
Steele PDB (latest 2.5 firmware)
FrSky X4r

I keep seeing "no connection to kiss fc" on the bottom of the screen after boot up. No amount of flashing or wiring has changed this. Is there an incompatibility issue with the V1 FC and the steele pdb? Have tried flashing 2.4, 2.5, and 2.5.1.  I flash the config tool first, then the OSD hex file, both with battery on and both port plugs removed from the FC.

 All wiring is done correctly. PDB is set for 7v vtx power also.

Ideas?

awolf78, 2018-10-06:
There is definitely no compatibility issue. Is the serial configuration on the KISS FC set to OSD?

Heligrom, 2018-10-08:
I ended up getting it to work. There were bent pins for the serial
connections from the pdb to the fc. Board was brand new also and had bent
pins.

Took 4 hours to figure that out.

Thanks for the response though! Loving the setup currently