awolf78/KISS_OSD #56: Setting VTX Power via OSD (Wolf PDB with KISS FC v1)

Oberlurch3000, 2018-10-22:
Hello,

I've got a strange problem with setting my vtx power via OSD.
So first:
My TBS Unify Pro is soldered to the VTX Smart Audio Pad and the Audio in Pad on the PDB.
All connections between the FC an the PDB are the two plugs PORT 1 and 2.
My KISS FW ist RC36i and my PDB is up to date too.

Now if I go to the section where I can set VTX Power, Channel and so on I get a "bug" at the power, where I can't set it higher. It seems that roll right doesn't work.
In the whole OSD everthing works fine, just at this point I gets stuck.

After some work on the drone I am able to set the power up to 200mW but not higher.
Downwards from 600mW or 800mW is alway possible... but setting it higher again not.

Therefore I have to use the KISS GUI and I can set it to 600mW or higher again.

Any advice would be great!

awolf78, 2018-10-22:
Did you unlock your VTX?

Oberlurch3000, 2018-10-22:
Yes I did. I can set it up to 800mW with KISS GUI but not via OSD.

awolf78, 2018-10-22:
What is your VTX max power set at?

Oberlurch3000, 2018-10-22:
Wow, just got it. Tried to make a short clip to show you and saw that there is a advanced setting with max power value... that solved my problem.
Thank you very much!

awolf78, 2018-10-22:
Great!

closed 2018-10-22 by awolf78