awolf78/KISS_OSD #58: Correct settings for rssi Steele pdb

Carmaniac626, 2018-12-08:
Hi there.
I've been following the few videos online and tutorials of how to setup the rssi to show up in my osd and so far I've gotten it to show up but I get 0-1db. Does that mean that perhaps my scale in the config tool isn't set correctly ? I have it currently set to 1009-2000 and the mixer on my taranis is weight 100 scale 0.

Tanks in advance

awolf78, 2018-12-08:
Did you double check the RSSI section here:
https://github.com/awolf78/KISS_OSD/wiki#setup-of-kiss-osd

Carmaniac626, 2018-12-09:
Hi there,

yes I did review it.

If i mess around with the weight and offset in the mixer page of my radio i
can see receiving signal but goes int negative and it doesnt make any sense.

https://newbeedrone.com/products/frsky-2-4ghz-xsr

Here is a link to the receiver I am using currently. In the forum they
mention to flash the firmware if the xm+ is being used. I didnt find that
necessary since im using a different receiver but maybe that is the case.
Could you confirm if that would be something for me to try or not ? I dont
want to try and flash my receiver and get it damaged.

I played around with the setting of the kiss osd and that didnt change
anything. The FC sees the input and reads it as a value between 1000-2000
ppm but I dont know where is the missing link. My radio sees the rssi. The
Flight controller does as well, however the osd can see it in negative
value if I do weight 200 and offset -100 but the numbers dont make any
sense.

Best regards,

Rumen

awolf78, 2018-12-09:
Can you post a DVR or video of what is happening?

Carmaniac626, 2018-12-11:
Yes I will try to do it today or tomorrow. Thanks for the prompt response.

- Rumen

TheRcCrawlerKing, 2019-01-06:
Is the Mr. Steele PDB capable of receiving RSSI data from a Spektrum 4649t receiver?