awolf78/KISS_OSD #61: No connection to kiss fc

Jeff37382, 2019-02-23:
Fresh build only hovered. At first I had reading in my osd when I was finishing. I go to actually fly and now at the bottom of my osd it reads "no connection to kiss fc" I've tried Reflashing, I've checkes the pins in the connector for damage. What's next?

awolf78, 2019-02-23:
Selected OSD in KISS GUI (not logger!)?
Connected RX to TX and vice versa (FC to OSD)?

Jeff37382, 2019-02-23:
Totally stumped sir, it previously worked a few Xs before. So I have the
fame exact build already - works fine - just trying to iliminate the
suspect I took and plugged in the new build pdb into my prior setups fc and
it still not connected to kiss fc  - I then took the existing build pdb and
plugged it into the new fc and still not connected to fc - I'm pretty lost
on this one......as far as your previous response where do I select kiss
osd in the gui (under serial port config - pic attached)

awolf78, 2019-02-23:
KISS GUI: Advanced tab, Logger config set to Logger disabled/OSD

Jeff37382, 2019-02-23:
Thanks for your help - found the issue - for whatever reason I'm not
getting power for the xtx on the steel pdb so I used the 7v common and the
5v pad just behind the vtx pad and had it bridges slightly. Fixed the joint
and no issues....... For now. Thanks again.

majestik801, 2019-09-09:
I’m having a similar issue. Everything looks normal and the issue just came about. The quad still flies completely smooth as hell and if there wasn’t a notification on my osd saying “Fc connection Lost” or “No connection to fc” I wouldn’t think anything was wrong. I changed the cables plugs and resoldered my rx.