awolf78/KISS_OSD #62: Steele pdb V1 impossible to update firmware

Parasitev, 2019-02-24:
Hi! I have 3 Aliens with the steele PDB, 2 with V1 and another V2. My problem is with one of the V1, when i try to update firmware from Xloader it just stay stuck on uploading...
The other two boards do it fast as always but this one is never passing the uploading status (i connect the lipo of course). I tryed from another computer, a laptop with the same result. I also tryed the Old School method with Arduino 1.8.8 and does same, after one minute it post "problem uploading to board" programmer is not responding. Seems something is fried? the quad fly perfectly and i can see Osd working but with old settings and version and i need to get RSSI working correctly. Anything i can do? Thanks in advanced.

awolf78, 2019-02-24:
I'm afraid there is not much you can do. Most likely an issue with the FTDI chip or the bootloader section in the ATMEL chip. Sorry.

Parasitev, 2019-02-24:
Thanks for your very fast responding, so if it is an issue with the FTDI chip or bootloader section.. Can i get a warranty replacement from ImpulseRC? should i writte them? It is very sad i cant change any settings on the OSD or config the RSSI channel correctly with my new crossfire system :(

awolf78, 2019-02-24:
Yes, you can contact your dealer for a replacement. Not sure how old the board is, so no guarantee that they will replace it.