awolf78/KISS_OSD #64: No Mah drawn on old

alexof92, 2019-09-07:
Hello, I've a problem with the Steele pdb, I don't have the current drawn don't work inside the osd. The esc are speedy 32 bit LS40A slim. I wired the telemetry and in kiss GUI I have the Voltage from each esc, the temperature, but not the Amps. Can someone help me?

awolf78, 2019-09-08:
Those ESCs have no current sensing, that's why you are not getting any amps.

LoKKeR96, 2019-11-26:
I have the same issue with KISS ESC 2-5S 25A (40A limit) 32bit and they do have the current sensing.
Any suggestion? Maybe the protocol used? (D-shot and so on)

awolf78, 2019-11-27:
@LoKKeR96 Looks like your ESC telemetry is not working. Take the props off and check in the KISS GUI if in the Data Output tab you can get a reading from each ESC for temperature, voltage, etc. (you can change the graph by selecting a different output on the drop down above the graph). If the KISS FC cannot read the ESC telemetry, the OSD won't be able to receive it either (it is looped through the FC).

LoKKeR96, 2019-11-29:
I got it working on the KISS FC v1. Not sure why it wasn't working on v2.