awolf78/KISS_OSD #65: All motors spinning in the same direction (right)

LoKKeR96, 2019-11-28:
I have just configured my KISS FC V1 with KISS 25A ESC and they all spin in the same direction. No matter the protocol or the yaw configuration. What's going on?

fedorcomander, 2019-11-28:
thats what they do. change it in esc flasher or solder jumper or swap 2 wires. 

---
Cheers,
Alex

LoKKeR96, 2019-11-29:
I soldered the jumper and that solved the problem but for some reasons on the kiss website (https://flyduino.gitbook.io/kiss-documentation/english-kiss-docu/kiss-esc-25a/esc-wiring-and-connections) it says: "This needs to be done only if you are NOT using DSHOT! " which kind of worries me. It is not very clear what they mean. If it's is bad or if you shouldn't need to do it.

closed 2025-08-31 by lcucurachi